tsandmann/freertos-teensy #1: How to add new libraries to project

MArndt26, 2020-11-08:
I am trying to use freertos with the SD.h library.  When I add the include statement I get the following error. 

<pre>
Compiling .pio/build/teensy41/src/main.cpp.o
src/main.cpp:23:10: fatal error: SD.h: No such file or directory
************************************************************
* Looking for SD.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SD.h"
* Web  > https://platformio.org/lib/search?query=header:SD.h
*
************************************************************
   23 | #include < SD.h>
      |          ^~~~~~
compilation terminated. 
</pre>

When I change the platform variable in platformio.ini to from `https://github.com/tsandmann/platform-teensy.git`
to `teensy` I fix this dependency issue, however, then all of the freertos libraries will have a similar error.  How can I set up my
project so that I can use all the needed libraries?

tsandmann, 2020-11-09:
First of all: to add a library, I use the `lib_deps` option in `platformio.ini`, e.g. `lib_deps = https://github.com/tsandmann/freertos-teensy.git, https://github.com/PaulStoffregen/SD.git`. I'm not sure, if the teensy SD library is registered in the platformio library system so that it could be found automatically. But maybe there's also a bug in the custom platform package (see below) I'm using here, will check this later.

To make FreeRTOS work on the teensy boards I had to adjust some minor parts of the Teensy Arduino core library. Therefore the custom platform setting is used (`platform = https://github.com/tsandmann/platform-teensy.git`). Furthermore it uses a more recent compiler toolchain (e.g. to support C++17) and a slightly modified Arduino core library (https://github.com/tsandmann/teensy-cores). The latter is needed because FreeRTOS needs some interrupt vectors that are also used by the Arduino core (SysTick, SVT, PendSV, EventResponder, Audio) to be set up for the RTOS. The modified core library supports these services by running them ontop of the RTOS.

Unfortunately the latest updates (some days ago) of the Teensy SD library cause some incompatibilities with the custom core library. Therefore using the SD library currently doesn't work. I'll release a fix for that in the next days to make `lib_deps = https://github.com/tsandmann/freertos-teensy.git, https://github.com/PaulStoffregen/SD.git` work again.

Thank you for reporting this issue! 😄

tsandmann, 2020-11-14:
I've fixed the incompatibilities.

And there is a new [example](../tree/master/example/sdfat) that shows the usage of the SdFat library.
I hope that helps?

closed 2021-01-17 by tsandmann