tsandmann/freertos-teensy #3: freertos-teensy porting for Teensyduino

gpb01, 2021-01-09:
Hi Timo,

following my issue on Phillip library, I move here to ask you if you think to make a version of your library that can be used in the Teensyduino environment. :-)

Thanks a lot in advance,

Guglielmo

tsandmann, 2021-01-12:
Hi Guglielmo,

yes, I will have a look at it.

Regards,
Timo

tsandmann, 2021-01-17:
There is a first test version available which can be used with Teensyduino. If you want to try it out:
 1. Download the library [here](https://github.com/tsandmann/freertos-teensy/releases/tag/v10.4.1_v0.4) as a zip archive.
 1. In Teensyduino select "Sketch -> Include Library -> Add .ZIP Library" and specify the downloaded zip archive.
 1. #include ["arduino_freertos.h"](https://github.com/tsandmann/freertos-teensy/blob/master/src/arduino_freertos.h) in your sketch.

However, there are currently the following limitations:
 - No support for C++'s `std::thread` (the compiler provided by Teensyduino is too old for this).
 - If the sketch (or any included library) uses the `EventResponder` [class](https://github.com/PaulStoffregen/cores/blob/bf413538ce5d331a4ac768e50c5668b9b6c1901f/teensy4/EventResponder.h#L64), the `EventResponder::attachInterrupt()` [variant](https://github.com/PaulStoffregen/cores/blob/bf413538ce5d331a4ac768e50c5668b9b6c1901f/teensy4/EventResponder.h#L111) must not be used, otherwise FreeRTOS will stop working. An update of the Teenys cores library is required to make this work.
 - [`configUSE_MALLOC_FAILED_HOOK`](https://www.freertos.org/a00110.html#configUSE_MALLOC_FAILED_HOOK) has no effect, `vApplicationMallocFailedHook()` will not be executed if `malloc()` fails.
 
I haven't done much testing so far as I don't use Teensyduino for my projects.

gpb01, 2021-01-17:
Hi Timo,
**THANKS A LOT** !!!  For sure i will do some tests and inform you of any problem. :-)

gpb01, 2021-03-17:
Hi Timo,
I tried you new release 10.4.1-6 in Teensyduino but, but, with this release, with the same program, I get some errors related to '__isb', '__dmb' and '__dsb' (_e.g. : error: '__isb' was not declared in this scope __isb();_).

Is it anything related to the compiler version present in IDE 1.8.13? is it possible to find a _detour_?

TIA

Guglielmo

gpb01, 2021-03-17:
... so, I have included these #define in portable/teensy.h:

#define __isb() __asm volatile("isb" ::: "memory")
#define __dmb() __asm volatile("dmb" ::: "memory")
#define __dsb() __asm volatile("dsb" ::: "memory")

and there are no more errors. Is this a correct "_detour_" or can it cause problems? 

TIA,

Guglielmo

tsandmann, 2021-03-17:
You're right, that's a bug with Teensyduino.
The suggested #defines should fix this, yes.

gpb01, 2021-03-17:
Hi Timo,
**THANKS** !!!

fralbo, 2022-08-24:
> No support for C++'s std::thread (the compiler provided by Teensyduino is too old for this).

Does it explain the problem I encounter in this [issue ](https://github.com/tsandmann/freertos-teensy/issues/11)?

tsandmann, 2023-04-01:
With the latest version available (v10.5.1_v5), there is now basic support for Teensyduino 1.58.

Please note, until this [Pull request](https://github.com/PaulStoffregen/cores/pull/683) is merged, the `EventResponder::attachInterrupt()` variant must not be used, otherwise FreeRTOS will stop working. 
See [Readme](https://github.com/tsandmann/freertos-teensy/tree/master#teensyduino-usage) for details of the limitations with Teensyduino.

Stevenlawrencehoriba, 2023-05-25:
On Platform IO how do I add TeensyDuino to the freeRTOS build, so i can get libraries for Ethernet and SD card libraries

tsandmann, 2023-05-27:
You can add FreeRTOS as a library to TeensyDuino (see [README.md](https://github.com/tsandmann/freertos-teensy/blob/master/README.md)) if you want to use it with TeensyDuino.
Or you can add Platform IO libraries as usual when using a Platform IO project.
But you can't use TeensyDuino together with Platform IO.

Stevenlawrencehoriba, 2023-06-01:
this is the issue, I need to be able to run the devices under the port of freeRTOS for teensy 4.1,  Your current  port will compile nicely with platformIO and i can add my own project to the port from the folder etc, and it works ok, until I start wanting devices. So its really the other way round I want the freeRTOs port , with access to the hardware devices, otherwise there is no point in using the freeRTOS port.  I need to use Ethernet on one task , serial devices on another task and business logic in between, using thrreads. as freeRTOS supports all the nice features you would expect, such as queusu and semaphores, it seems natural that tis is a basic requirement to also access the internal devices, whioch are linked to you locked source for Arduino, unless you know of another way?

Stevenlawrencehoriba, 2023-06-01:
further info today.,. I have got freeRTOS for teensy working, using version 10.5.1_v6 with QNEthernet on one thread and LED on another. We need serial and sdcard threads. Do you think if these are isolated on there own threads and messages sent across to those threads to action device functions this would be OK. I had to use the URL method of adding in the libraries by hand as required. The issue i guess is going to be IF the libraries calsl something that upsets the RTOS.

Stevenlawrencehoriba, 2023-06-02:
additional information today ongoing work trying to get device to work on your freeRTOS port. Although the Ethernet server works  in its own thread, adding the SDFat library is now an issue. In that If I try and initialise SDFat with sd.begin(SdioConfig(FIFO_SDIO), where it is a global, it will not initialise, If i place the "SdFat32 sd;" object inside the start of a task i.e. beginning of a task thread, it will initialise, but fail to work, i.e. any file control fails. I would be happy to share my entire project if that would help, as it really is simple test code , the same code in the thread,  i have created a std arduino sketch under the Arduino IDE , and it works fine. So it MUST be something about you port of freeRTOS, and interaction with your devices. What or how can i do to help with getting the device to work under freeRTOS?

tsandmann, 2023-06-02:
I did test the SDFat library as shown in the sdfat example, but not beyond that. So yes, there may be issues with other libraries, especially if they are not written for FreeRTOS. Do Ethernet and SDCard on Teensy use the same hardware controller or something? Maybe there is an interference or a hardware access is preempted when it shouldn't be. 
The FreeRTOS port doesn't contain specific drivers for the Teensy, so your question goes far beyond this issue here.

Stevenlawrencehoriba, 2023-06-02:
tonights findings. Finally got SDFAT working.. this may help other people..  make sure SDFat32 is initialied at the start of a thread. and to make the file read and writes work, use the defines as follows, not FILE_READ and FILE_WRITE

 file = psd->open("hi.txt", O_WRITE | O_CREAT | O_TRUNC);

file = psd->open("hi.txt", O_READ);

as the defines FILE_READ and FILE_WRITE clash with another library "FD.h" ?

tsandmann, 2025-06-08:
Hi,

It's using malloc()/free() provided by the c-library that is used by the teensy library, very similar to the Heap_3 option of FreeRTOS.

I've tested the blink and sdfat examples from the library with Teensyduino 1.59 on a Teensy 4.1.

npinjala, 2025-06-15:
Thank you, Timo.