tsandmann/freertos-teensy #6: Platform IO SD Card Example

MichaelBMiner, 2021-09-24:
Hello,

I am trying to use your SD card example. However I got the error that FileImpl was not found or defined. Turns out PlatformIO for VSCode uses its own libraries rather than those used by Arduino and Teensyduino. 

I was wondering do you know how to point my application to use the correct libraries? Or how do you run this? I am fine not using PlatformIO, I just want to get it running.

tsandmann, 2021-09-24:
You'r right, the SD card example isn't working any more. This is because it references the latest version of the SD library which was updated recently and unfortunately this update brought some incompatibilities. I will look into it and fix this incompatibility soon. 

As a workaround, you can use the old version of the library by changing the `platformio.ini` file of the example as follows:

```patch
- lib_deps = https://github.com/tsandmann/freertos-teensy.git, https://github.com/PaulStoffregen/SD.git, https://github.com/PaulStoffregen/SdFat-beta.git, https://github.com/PaulStoffregen/SPI.git
+ lib_deps = https://github.com/tsandmann/freertos-teensy.git, https://github.com/PaulStoffregen/SD.git#0eccd950000a456bd886f390130c205fbfbbc0c8, https://github.com/PaulStoffregen/SdFat-beta.git, https://github.com/PaulStoffregen/SPI.git
```

In short: append `#0eccd950000a456bd886f390130c205fbfbbc0c8` to `https://github.com/PaulStoffregen/SD.git`.

I'm using PlatformIO for my projects. There is also some basic support for Teensyduino, see [#3](https://github.com/tsandmann/freertos-teensy/issues/3#issuecomment-761828381) for details. As I'm using PlatformIO with VSCode, there isn't that much testing with Teensyduino done by me. Therefore it might not work with every version, etc.

MichaelBMiner, 2021-09-24:
When you run your example. In VS Code should I just navigate to the blinky folder and open that as the PIO project? Everything should work from there?

tsandmann, 2021-09-24:
> When you run your example. In VS Code should I just navigate to the blinky folder and open that as the PIO project? Everything should work from there?

Yes.

tsandmann, 2021-09-25:
The incompatibilities with the SD library are fixed now. You need to run `pio update` to update the libraries.

closed 2021-09-25 by tsandmann