tsandmann/freertos-teensy #8: Teensy 4 compile error due to missing symbol

arandomliz, 2021-11-09:
From cd38223090558cc0ef00c17b749277320f105801 onward the compiler will throw:
```
Linking .pio/build/teensy40/firmware.elf
/home/elizabeth/.platformio/packages/toolchain-arm-cortexm-linux/bin/../lib/gcc/arm-cortexm7f-eabi/11.1.0/../../../../arm-cortexm7f-eabi/bin/ld: .pio/build/teensy40/libb49/libfreertos-teensy.a(teensy_4.cpp.o): in function `vPortSetupTimerInterrupt':
teensy_4.cpp:(.flashmem+0x25c): undefined reference to `unused_interrupt_vector'
collect2: error: ld returned 1 exit status
*** [.pio/build/teensy40/firmware.elf] Error 1
```

I assume it is due to the introduction of `unused_interrupt_vector` as an external void in the file `src/portable/teensy_4.cpp` without mention it anywhere else in the code except in `vPortSetupTimerInterrupt`

tsandmann, 2021-11-09:
Are you using the latest version of the teensy core framework? Could you run `pio update` (or `pio update --dry-run`) to check if there are any outdated libraries in use?

`void unused_interrupt_vector(void)` was a static function of the framework startup code in an earlier version which is incompatible now with latest freertos-teensy. 

Maybe it's the same for #7?

arandomliz, 2021-11-09:
Well. It still seems to be mentioned in the current version of the code:
https://github.com/tsandmann/freertos-teensy/blob/c0ff2e5e25514b7aa131be2094f841ab96d5019b/src/portable/teensy_4.cpp#L243

tsandmann, 2021-11-09:
Yes, that's fine, because `void unused_interrupt_vector(void)` isn't declared static any more in the latest framework version:
https://github.com/PaulStoffregen/cores/blob/2e9aa7a34f2e7914767fde52fd1a9c3be7c30dd1/teensy4/startup.c#L527

As I can't reproduce your issue, could you please tell me the version of the package `framework-arduinoteensy` you're using?

arandomliz, 2021-11-09:
Well that's interesting. I'll do so later, but it should be about 3 days old at most. At least that is when I did a `pio init` on one of the projects that threw this error.

arandomliz, 2021-11-10:
Okay. It does seem like that you are right and the problem resolves itself with a `pio update`.

If others hit this: A simple `pio update` wasn't the solution for me. The update process crashed whilst pulling the tsandmann/platform-teensy. I needed to delete the `teensy@...` platform in `~/.platformio/platforms`, install the teensy platform again and was only then able to pull the latest freertos platform-teensy using `pio run` and `pio update` afterwards

closed 2021-11-10 by arandomliz

tsandmann, 2021-11-10:
> If others hit this: A simple `pio update` wasn't the solution for me. The update process crashed whilst pulling the tsandmann/platform-teensy. I needed to delete the `teensy@...` platform in [...]

Thanks for the hint on this 🙂