tsandmann/freertos-teensy #11: analogWriteResolution(12), freezes the task and stop the PWM

fralbo, 2022-08-23:
Hello,
Teensy 4.0
TeensyDuino 157
Frequency 600 MHz

I encounter a problem using PWMServo and investigating in the code, I discovered that the call to `analogWriteResolution(12)`causes the calling task to freeze and the PWM to stop. Additionnaly, the orange LED is blinking 4 times every 4 sec. Don't what could it mean.

Here is my code:

    #include <Arduino.h>
    #include "arduino_freertos.h"
    #include <PWMServo.h>

    #define Serial Serial

    #define SERVO_PWM              2
    #define CYCLE_TIME                3

    PWMServo myServo;
    int8_t cycle;

    static void task_ICM( void *) {

      for (;;) {
        cycle = !cycle;
        digitalWriteFast(CYCLE_TIME, cycle);

        uint32_t oldres = analogWriteResolution(12);
        analogWrite(SERVO_PWM, 301);
        analogWriteResolution(oldres);

        ::vTaskDelay(1);
      } // For ever   
    }

    void setup()
    {
      pinMode(CYCLE_TIME,arduino::OUTPUT);  //cycle time measure
      pinMode(SERVO_PWM,arduino::OUTPUT);  //Servo PWM
      
      Serial.begin(115200); // Start the serial console
      while (Serial.available()) // Make sure the serial RX buffer is empty
        Serial.read();

      delay(100);

      // Initialize Servo
      myServo.attach(SERVO_PWM);
      Serial.println("Calling Servo.write()");
      myServo.write(90); // Works fine here
      
      Serial.println();
      Serial.println(PSTR("Configuration complete!"));

      arduino::Serial.println(PSTR("\r\nrunning FreeRTOS kernel " tskKERNEL_VERSION_NUMBER "."));
      ::xTaskCreate(task_ICM, "task_ICM", 2048, nullptr, tskIDLE_PRIORITY + 0, nullptr);
      ::vTaskStartScheduler();
    }

    void loop()  { }

I already looked for info on FreeRTOS forum without success.
Any idea about what could cause this problem?

tsandmann, 2022-08-24:
I did a quick test of your code and the problem only occurs with Teensyduino. It seems to be an incompatibility with Teensyduino's core library. It will be fixed in the next release of this library.

As a workaround, you can test the following change to `src/portable/teensy_4.cpp`: add before line 257: 

```c++
_VectorsRam[0] = reinterpret_cast<void (*)()>(&_estack);
```

so that the whole block looks like:
```c++
/* override arduino vector table entries */
_VectorsRam[0] = reinterpret_cast<void (*)()>(&_estack);
_VectorsRam[11] = vPortSVCHandler;
_VectorsRam[14] = xPortPendSVHandler;
_VectorsRam[15] = xPortSysTickHandler;
```

That should resolve the problem; if not, please let me know.

> Additionnaly, the orange LED is blinking 4 times every 4 sec. Don't what could it mean.

4 blinks indicate a hard fault exception. 3 blinks: a stack overflow was detected, 1 blink: `assert()` was called.

fralbo, 2022-08-25:
Hello @tsandmann ,
Many thanks for your reply, I'll give a test ASAP.
Just a question; Are the LED status a Teensyduino implementation?

tsandmann, 2022-08-25:
> Just a question; Are the LED status a Teensyduino implementation?

Teensyduino had that for hard faults in an older version. Now it reboots after a few seconds in this cases and creates a [crash dump](https://www.pjrc.com/teensyduino-1-54-released/).

commit 01c1d397 (2022-08-25): ensure correct vector table setup for stackpointer

fixes #11
closed 2022-08-25 by tsandmann