tsandmann/freertos-teensy #12: Unaligned access UsageFault

gertorro, 2022-09-05:
Hello, I am trying to use your port of freertos to teensy to build an acquisition system which communicates over Ethernet.
I was able to implement many functionalities without any issue. I managed to get UCP communications working without problems, analogRead from the ADC and a simple state machine, all working in separate tasks. I need to read a device attached to the SPI bus, I have tested the read/config code without freertos withou any issues. It also works in the setup() function but the moment I put the read code inside a task I get the following error:

I/O configured and initialised
Starting LTC2983
Setup of  LTC2983 complete
readADC channel 1 state=5 val=-273.15
Ethernet cable is not connected.
Ethernet initialised
running FreeRTOS kernel V10.4.6.
setup(): starting scheduler...
Fault IRQ:    0x00000006
 sp:          0x2021E5B0
 lr:          0xFFFFFFED
 stacked_r0:  0x00000000
 stacked_r1:  0x00000000
 stacked_r2:  0x00000001
 stacked_r3:  0x403A0000
 stacked_r12: 0xA5A5A5A5
 stacked_lr:  0x0000B831
 stacked_pc:  0x0000C4D4
 stacked_psr: 0x41000000
 _CFSR:       0x01000000
  (UNALIGNED)   Unaligned access UsageFault
 _HFSR:       0x00000000
 _DFSR:       0x00000000
 _AFSR:       0x00000000
 _BFAR:       0x00000000
 _MMAR:       0x00000000
Active task (TCB): readTCfro
Active task (stack): readTCfro

And the system keeps throwing memory management errors at me. 
Here is the task's code:
static void readTCfromLTC2983(void*)
{
  while (true)
  {
    ltc.beginConversion(1);
      do {
         ::vTaskDelay(pdMS_TO_TICKS(1'0));
      } while (!ltc.isDone());
      double val = ltc.readTemperature(1);
      Serial.print("readADC channel 1 state=");
      Serial.print(ltc.getState());
      Serial.print(" val=");
      Serial.println(val);
    Serial.println("Reading temps");
    ::vTaskDelay(pdMS_TO_TICKS(500));
  }
}

I have also tried to access directly the SPI  directly without the library for the LTC2983 from inside the task without success. In this case the task is written as:
static void readTCfromLTC2983(void*)
{
  while (true)
  {
      SPI.beginTransaction( SPISettings(2000000, arduino::MSBFIRST, SPI_MODE0));
      digitalWrite(10, arduino::LOW);
      SPI.transfer(LTC298X_SPI_WRITE);
      SPI.transfer16(LTC298X_ADDR_CMD);
      SPI.transfer(LTC298X_CMD_BEGIN | 1);
      digitalWrite(10, arduino::HIGH);*/
    ::vTaskDelay(pdMS_TO_TICKS(500));
  }
}
And the errors I get are these:
static void readTCfromLTC2983(void*)
{
  while (true)
  {
    //ltc.beginConversion(1);
    //Test barebones:
      SPI.beginTransaction( SPISettings(2000000, arduino::MSBFIRST, SPI_MODE0));
      digitalWrite(10, arduino::LOW);
      SPI.transfer(LTC298X_SPI_WRITE);
      SPI.transfer16(LTC298X_ADDR_CMD);
      SPI.transfer(LTC298X_CMD_BEGIN | 1);
      digitalWrite(10, arduino::HIGH);
    /*ltc.beginConversion(1);
      do {
         ::vTaskDelay(pdMS_TO_TICKS(1'0));
      } while (!ltc.isDone());
      double value = ltc.readTemperature(1);
      Serial.print("readADC channel 1 state=");
      Serial.print(ltc.getState());
      Serial.print(" val=");
      Serial.println(value);
    Serial.println("Reading temps");*/
    ::vTaskDelay(pdMS_TO_TICKS(500));
  }
}

I/O configured and initialised
Starting LTC2983
Setup of  LTC2983 complete
readADC channel 1 state=5 val=-273.15
Ethernet cable is not connected.
Ethernet initialised

running FreeRTOS kernel V10.4.6.
setup(): starting scheduler...
Fault IRQ:    0x00000005
 sp:          0x202206D8
 lr:          0xFFFFFFFD
 stacked_r0:  0x0000000A
 stacked_r1:  0x00000000
 stacked_r2:  0x00000001
 stacked_r3:  0x403A0000
 stacked_r12: 0xA5A5A5A5
 stacked_lr:  0x00000259
 stacked_pc:  0x0000C58C
 stacked_psr: 0x21000000
 _CFSR:       0x00008200
  (PRECISERR)   Data bus error (address in BFAR)
  (BFARVALID)   Bus Fault Address Valid
 _HFSR:       0x00000000
 _DFSR:       0x00000000
 _AFSR:       0x00000000
 _BFAR:       0x403A002C
 _MMAR:       0x403A002C

Active task (TCB): readTCfro
Active task (stack): readTCfro

What could be causing this behaviour? Thanks in advanced!

tsandmann, 2022-09-06:
Looks like some data structures of the SPI driver get corrupted. Maybe the stack size of the task is too small? Or SPI is also used from another task? 
I don't think it's an issue with the FreeRTOS port.

gertorro, 2022-09-07:
I managed to get it working by changing the optimization configuration of teensyduino from "Fast" to "Fastest".
No issues with the SPI whatsoever and the code works like a charm. Thanks for the wonderful port @tsandmann !!

closed 2022-09-07 by gertorro