tsandmann/freertos-teensy #14: Issue with using PlatformIO

windorey, 2023-01-14:
Error when compiling blink example:

```
.pio/build/teensy41/libFrameworkArduino.a(startup.c.o): In function `_sbrk':
startup.c:(.text._sbrk+0x0): multiple definition of `_sbrk'
.pio/build/teensy41/libc07/libfreertos-teensy-master.a(teensy_common.cpp.o):teensy_common.cpp:(.text._sbrk+0x0): first defined here
/home/kemal/.platformio/packages/toolchain-gccarmnoneeabi@1.50401.190816/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Disabling relaxation: it will not work with multiple definitions
collect2: error: ld returned 1 exit status
*** [.pio/build/teensy41/firmware.elf] Error 1
===============================================
```

tsandmann, 2023-01-14:
Did you modify the [platformio.ini file](../blob/2634478ffd9d42e377b72e0ff118755b5088058d/example/blink/platformio.ini) of the example? The error looks like an incompatible platform reference.

windorey, 2023-01-14:
> Did you modify the [platformio.ini file](../blob/2634478ffd9d42e377b72e0ff118755b5088058d/example/blink/platformio.ini) of the example? The error looks like an incompatible platform reference.

No, I did not. Thank you.

closed 2023-01-14 by windorey