tsandmann/freertos-teensy #19: Latest PULL example stdThread Hangs at runtime

Stevenlawrencehoriba, 2023-05-22:
commenting out task3 fixes the issues, assume this is being tested for std:async and futures. Also the serial port has been set in the code to 115200, but platformIO.ini has not set the serial port up to 115200, so default to 9600 baud

tsandmann, 2023-05-27:
What hardware are you using? The example works for me as expected on a Teensy 4.1.

The examples uses an USB emulated serial device which doesn't use the baud rate parameter at all. Therefore, the value doesn't matter.

Stevenlawrencehoriba, 2023-06-01:
when using a physical teensy 4.1 it fails to run on the teensy, from platformio running the serial console
.

tsandmann, 2023-06-02:
What output do you get on the serial console?

tsandmann, 2023-10-07:
Issue closed because it's stale.

closed 2023-10-07 by tsandmann