tsandmann/freertos-teensy #23: teensy-FreeRTOS event group handler Issue when trying to use C++ Exceptions

yamaan93, 2023-10-16:
I have been trying to add a try-catch statement to my project, so I added `-fexceptions` and `-xc++` to my build flags to enable that functionality. However when I do that I get a compiler error from teensy-freertos:

```
error: invalid conversion from 'void*' to 'EventGroupHandle_t' {aka 'EventGroupDef_t*'} [-fpermissive]
```

Not sure what exactly this means but I haven't found anything online that would suggest that this is a larger freeRTOS issue as people have been able to enable exceptions in other vanilla freertos projects, so I thought I'd raise it here.

tsandmann, 2023-10-16:
You can't build FreeRTOS with `-xc++`. 
However, I don't think you need `-xc++` to enable exception handling.

closed 2023-10-16 by tsandmann

yamaan93, 2023-10-17:
Without `-xc++` you end up with the following: 

```
C:\Users\Yamaan\AppData\Local\Temp\ccEqBZHb.s:34: Error: symbol `__cxa_end_cleanup' is already defined
lto-wrapper.exe: fatal error: arm-cortexm7f-eabi-g++ returned 1 exit status
```
This probably isn't freeRTOS related but I thought I'd put it here in case anyone else looks in the future.