tsandmann/freertos-teensy #25: A thank you and a question.

g40, 2024-04-13:
Hello Timo

Many thanks for this excellent work. 

Can I make a very small set of contributions in return? Would you be interested in a PR or two for extended examples? I've got networking just about running on a T4.1 using the ethernet adapter and a dedicated FreeRTOS task using PlatformIO.

Bets wishes,

Jerry.

tsandmann, 2024-04-14:
Hi Jerry,

Thank you very much!
Yes, of course, I'm always happy to get extended examples. 👍

g40, 2024-04-18:
Hi there Timo. 

It looks like you made a (very) good number of commits to the Teensy-cores code. Do you think it is realistic to be able assemble patches so the Teensy runtime can be kept up to date with the PJRC repo? I'm just wondering if it can be automated.

Also, I am working on getting very basic GDB support integrated. I'm not a low-level ARM expert, so another question is this; Can I use 
`IRQ_SOFTWARE` for my debug software interrupt? And is it OK to keep using the `IntervalTimer` which runs off `IRQ_PIT`?

Many thanks.

tsandmann, 2024-04-24:
Hi Jerry,

The changes to teensy-cores needed for FreeRTOS are only those in [this pull request]( https://github.com/PaulStoffregen/cores/pull/683). All other changes are just to support a more recent compiler (because I like to use features of a more recent C++ standard). 

Having basic GDB support is a great idea, I haven't found the time to look into it yet, but would really appreciate it.
`IRQ_SOFTWARE` isn't used by FreeRTOS, but it's used by the audio classes from [teensy-cores](https://github.com/PaulStoffregen/cores/blob/02944ba831f2804bdffa345dfa86a63ec652f4f9/teensy4/AudioStream.h#L169). 
If you're running FreeRTOS, you can put the audio processing in a high-priority FreeRTOS task (I've done this without any problems), but you would need to patch Teensy's audio code to not use `IRQ_SOFTWARE` for the task activation then.

Using `IntervalTimer` should be fine, I've never used it myself, but `IRQ_PIT` isn't used by the FreeRTOS port at all. You may need to check the ISR priorities, if you need to call FreeRTOS API functions from these ISRs (or if you want them not to be masked by FreeRTOS critical sections).

g40, 2024-04-24:
Hello Timo,

Great news, thank you. I will make certain of the audio aspect as that is my core interest.

I've got a GDB stub running on the T41, the idea being you simply `#include gdbtx.h` and then connect with GDB from the host. Being RTOS aware would be extremely useful. If you are interested I'll ping you when it is slightly less of an ARM assembler lesson.  Your support and knowledge would be invaluable.

Jerry

g40, 2024-04-24:
Timo, forgot to ask, 

Did your PR ever make it into teensy-core? From here it looks like it was forgotten?

tsandmann, 2024-04-24:
Hi Jerry,

> I've got a GDB stub running on the T41, the idea being you simply `#include gdbtx.h` and then connect with GDB from the host. Being RTOS aware would be extremely useful. If you are interested I'll [...]

Oh yes, I'm definitely interested, thanks!

> Did your PR ever make it into teensy-core? From here it looks like it was forgotten?

No, it hasn't been merged into teensy-core yet.