tsandmann/freertos-teensy #29: https://github.com/tsandmann/platform-teensy.git Old Verion Question

Stevenlawrencehoriba, 2024-08-05:
Hi Tim, can you provide me with a link so i can fix the build to an older version, that was released with version 10.5.1. FreeRTOS. I need to check if this works, if i use the link above it wil just pull the latest. Which we dont want. I can manage the rtos version manually. I woudl like to try the older verion. Thanks

Stevenlawrencehoriba, 2024-08-08:
thanks for the info, but that does not hewlkp us either , some other things have totoally broken our code..

tsandmann, 2024-08-10:
The versions of the platform library and this FreeRTOS port are independent, there is no 'released with version X'.
If you don't need support for the `EventResponder` class (see Readme), you can use the original teensy platform (`platform = teensy`) from PlatformIO instead of the adapted one to rule out that the issues are related to this.

If you're getting asserts from FreeRTOS, there's probably something wrong with your application code or the way you're using the FreeRTOS API. But that's definitely out of scope for this port of FreeRTOS for the teensy boards. I'm happy to try to help with issues related to the teensy adaptions, but for general issues realted to the RTOS itself, the official FreeRTOS support options are definitely the better place to go.

closed 2024-08-10 by tsandmann