tsandmann/freertos-teensy #31: Heap Size Questions

Stevenlawrencehoriba, 2024-09-24:
So after a lot more investigations and time, since the updates I added a monitor task to dump various freeRTOS system values. the free heap used to be in order of a few hundred, and since the update without any networking task running, is now 28?. The returned value from xPortGetFreeHeapSize(); doesnt state if that is bytes or kbytes, I can only assume its bytes, with the unpredictable behavours I am getting. So what can be done.. can i increae the heap size as per FreeRTOS docs, for a test, and if so what is the max it can be set at, What is the current default size. More importantnly , what's happened to all the free memory since the updates of core 1.59, compilier/linker and FreeRTOS. Has there been any changes here with this port? This is costing us an imeserable amount of time to resolve, causing budget over spends etc.. I have tried all the optimising choices, and stil get a low Free HEAP with some tasks runing, adding one more task or even the initialisation of that task, just causes the system to constantly re-boot. Interestignly , if left just sat re-booting, in this case, the system suddenly fire s up, but the heap is way too low.

We use the EXPRAM , (both chips) can the heap be moved to use this RAM instead, I accept the performance hits. What needs to be changed to allow this instaead of RAM2.

Stevenlawrencehoriba, 2024-09-24:
on a build its not using massive amount of memory
RAM 1:  [===       ]  32.3% (used 169344 bytes from 524288 bytes)
RAM 2:  [          ]   2.8% (used 14476 bytes from 524288 bytes)
Flash:  [          ]   3.8% (used 307900 bytes from 8126464 bytes)

tsandmann, 2024-09-28:
This port directly used the heap that's provided by libc (otherwise C++ couldn't be used sensibly).

`xPortGetFreeHeapSize()` just returns `fordblks` from libc's `mallinfo()`.
Most libc's heap implementations have an initial heap size of 0 when the system/program is started and increase the heap size on demand during runtime. To get additional information about the heap usage, you can call [`mallinfo()`](https://linux.die.net/man/3/mallinfo). 
A low return value of `xPortGetFreeHeapSize()` does **not** necessarily mean that there's no more memory available, it just means that the next heap memory allocation request requires a call to `_sbrk()` to increase the heap size.

This port uses all free memory in RAM 1 for libc's memory requests. If you want to use other memory for the heap, you'd need to provide your own implementation of `_sbrk()` and link against that.

closed 2024-10-29 by tsandmann