tsandmann/freertos-teensy #34: teensy4.1 USB Serial not working in setup()

rmx-it, 2025-05-27:
Hi,
is it intentional that Serial isn't working during `setup()`?
`/dev/ttyACM0` seems not to be available during `setup()` (starts after `::vTaskStartScheduler();` ?)
A  `while (!Serial)` in  `setup()` hangs forever.

A `Serial.print()` in tasks is running fine.

Is this due to not initializing USB early.
Could this be fixed that at least USB Serial is working during `setup()` (for debugging purposes)?

Env: Ubuntu, teensy4.1, current platformio, current freertos-teensy.

tsandmann, 2025-06-08:
Hi,

The way FreeRTOS is designed, it expects that no interrupts are used before `vTaskStartScheduler()` is called. Because USB Serial requires interrupts, it doesn't work during `setup()`. 
I was thinking about a workaround for that some time ago, but haven't tested it yet. It might be supported with the next release.

tsandmann, 2025-06-10:
You can try out the branch [v11.2](/tsandmann/freertos-teensy/tree/v11.2) if you like to test it.

rmx-it, 2025-06-16:
Hi,
that with the non-working interrupts was my assumption too but otherwise even in the blink example Serial
is user in setup()....

with the 11.2 branch this works now:
...
:Serial.begin(115'200);
  while (!::Serial) {
    yield();
  }
  clearSerialInput();
  Serial.println("Start");
...

Thanks!

closed 2025-06-20 by tsandmann